TinyTapeout/tt123-demo-pcb #2: Add pokey holes

mattvenn, 2023-08-27:
Removing the breakout is very hard, add pokey hole to push it out from the back

psychogenic, 2023-08-28:
Hm, could you describe your vision/requirements for this pokey hole?  This is a 2-layer board and the Pico is right under there... so what's a pokey hole that's better than just levering the thing out, but is doable without too much disruption?

mattvenn, 2023-08-29:
updated title and comment.
the issue atm is that you can't flash it with the board in place, so every time I tried new firmware I had to lever it out
I bent the pins after a few levering outs

psychogenic, 2023-10-11:
There's no good way to do this... I think it matters little, now that we can program the flash in place through HK SPI.  Suggest we close.

closed 2023-10-11 by mattvenn