TinyTapeout/tt123-demo-pcb #14: do we need an 'enable clock divider' toggle

mattvenn, 2023-10-21:
at the moment, you have to put a jumper between clkdv and 3.3v to enable it

mattvenn, 2023-10-22:
and would be good to visibly see it easily

mattvenn, 2023-10-23:
can see it with a firmware change to show it on the management gpio

psychogenic, 2023-10-24:
You're thinking just a 2-pin jumper to high for this?  And visibility: can we pwm the gpio pin at the rate of that clock, or something like that?

mattvenn, 2023-10-24:
yes exactly, just a 2 pin jumper. ATM I have to have a wire handy to connect the clkdiv pin to 3.3v. Something like what we have for the scan chain drive select.

I've already got a firmware that shows the frequency on the gpio led. So that's already working.

psychogenic, 2023-10-27:
Have this as a jumper, next to manual clock MOM.
[image]

closed 2023-11-09 by mattvenn