TinyTapeout/tt123-demo-pcb #15: change oscillator to 20M?

mattvenn, 2023-10-21:
needs testing first

psychogenic, 2023-10-24:
We can test by feeding in 20M and changing the clock select jump.  But did we need something with a funny duty cycle for this to work?  If so, would that be true for TT3 as well?

mattvenn, 2023-10-24:
Yes, we did test this a bit in the bringup. TT3 has a different chain that should mean we can just use an oscillator. 
I'll check how fast we can go with a 50% duty cycle and then make a call on the oscillator for TT2

mattvenn, 2023-11-09:
tested up to 28M, so let's go with a 20M oscillator

https://discord.com/channels/1009193568256135208/1162346245088608366/1166759290665705485

psychogenic, 2023-11-13:
Ok, done, changed for a 20M WE osc with same footprint.

psychogenic, 2023-11-13:
Changed to 830208220401, another Wurth IQXO-791 series osc.  It's the hi-fi version with the same footprint (only the best for my TT friends) but the 831056293 could probably do the trick as well, for a few pennies less.

closed 2023-11-13 by psychogenic