TinyTapeout/tt123-demo-pcb #18: feedback on v2.2.3

mattvenn, 2023-10-25:
DNP memory programming header
7 seg not soldered properly on 2 of 4 boards

[image]

psychogenic, 2023-10-27:
Have set the DNPs at Aisler.  Have the reset RC cap solution in latest rev, as well as the ~rst naming on the symbol.

psychogenic, 2023-11-13:
Headers J1, J3 and J9 are DNPed and excluded @ aisler.  WE 7-seg footprint was different enough to warrant a new one, which I created and populated. nRST cap fix applied, and added a DNP pull-up in case we find the internal isn't around or doing the job.  Submitted as revision 11 @ aisler, closing issue.

closed 2023-11-13 by psychogenic