TinyTapeout/tt123-demo-pcb #25: feedback on the pinout doc

mattvenn, 2024-03-08:
contrast of the 7-seg could be improved. 

7 seg looks a bit weird
[image]

Follow branding guidance (TinyTapeout -> Tiny Tapeout) (saw a ref to tinytapeout in scanchain control)

Would re-write this: 
[image]

to:

The clock and reset buttons are debounced switches connected to inputs 0 and 1 respectively. Not all projects use this convention, but for those that do, these are convenient ways to manually single step and reset.

would rewrite the 40M clock part of this:
[image]

to:

which provides a 40MHz clock (20MHz for TT02 demoboards)

Rewrite this:
[image]

To use this, set the divisor with the input switches and then connect the slow clk jumper. The slow clock will automatically be provided on input 0. 

Hard to fit in, but a few examples of slow clock settings would be nice, like the min and max :
all off = 2.3kHz
all on = 9Hz

Rewrite 'as fast as scanchain can manage' here: 
[image]

To: 10kHz for TT03 and 5kHz for TT02.

run a spellcheck on the text. I saw a few: seperation -> separation. there might be more.

mattvenn, 2024-03-08:
Also please add the pinout diagram into this repo

psychogenic, 2024-03-08:
All changes made, hacked up 7-seg image, pinout uploaded and in readme.

psychogenic, 2024-03-08:
Done

closed 2024-03-08 by psychogenic